mikaelpatel/Arduino-GPIO #8: In /src/Button.h millis() is stored in a 16 bit variable

JoakimCh, 2018-01-05:
To me it seems like an obvious error that the return value of millis() is stored in a 16 bit variable when the return value actually is 32 bits. This will lead to wrong behavior when it overflows.

mikaelpatel, 2018-01-06:
@JoakimCh This might seem strange but cutting an unsigned integer by dropping higher bits actually works just fine as it is compared to the debounce limit (which also is 16-bit). The wrap-around is in the range of a second. 

Having said all this we should construct a test case that shows that it could fail or not. Any suggestions?

JoakimCh, 2018-01-08:
Aha, I didn't think that far. Sound reasonable now when I think about it! I guess no test will be needed.

closed 2018-01-08 by JoakimCh