mikaelpatel/Arduino-GPIO #12: Unit Test / Mock

ellisgl, 2019-04-05:
Since this is a class, why no unit tests? Would be nice to be able to mock this up and unit test code against it.

mikaelpatel, 2019-04-05:
@ellisgl 
Thanks for your interest in this project. Could you give an example of unit test code that you think would be relevant? Typical code that would use the mock would also contain timing (for instance any of my classes that use GPIO). This is a great challenge to provide a mock environment for. 

In any case please add more info so I better understand your idea. 

Cheers!

ellisgl, 2019-04-05:
Example: I would like to unit test the SRLatch object.

`#include "GPIO.h"

class SRLatch
{
    protected:
        // Inputs
        GPIO _s; // Set pin.
        GPIO _r; // Reset pin.

        // Input type
        bool _inverted;

        // Outputs
        GPIO _q; // Normal.
        GPIO _nq; // Inverted.
        
        bool _state;
    
    public:
        SRLatch(GPIO& s, GPIO& r, GPIO& q, GPIO& nq, bool inverted);
        boid update(void);
};

SRLatch::SRLatch(GPIO& s, GPIO& r, GPIO& q, GPIO& nq, bool inverted = false)
{
    _s = s;
    _r = r;
    _q = q;
    _nq = nq;
    _inverted = inverted;
    _state = false;
}

void SRLatch::update(void)
{
    if(_inverted) {
        if (!_s && _r && !_lastState) {
            _q = HIGH;
            _nq = LOW;
            _state = true;
        } else if (_s && !_r && _lastState) {
            _q = LOW;
            _nq = HIGH;
            _state = false;
        }
    } else {
        if (!_s && _r && !_lastState) {
            _q = LOW;
            _nq = HIGH;
            _state = false;
        } else if (_s && !_r && _lastState) {
            _q = HIGH;
            _nq = LOW;
            _state = true;
        }        
    }
}

GPIO<BOARD::D0> s;
GPIO<BOARD::D1> r;
GPIO<BOARD::D2> q;
GPIO<BOARD::D3> nq;

SRLatch srlatch(s, r, q, nq, true);

void setup()
{
    s.input();
    r.input();
    q.output();
    nq.output();
}

void loop()
{
    srlatch.update();
}
`

mikaelpatel, 2019-04-06:
@ellisgl 
Ok. Now you need to explain what you mean with unit test. A mock GPIO class can give you the interface (API) but how about the semantics. How to you intend to check that? Traditionally unit test will feed some stimuli (data) and then check some results. Also what target/host do you want to run the unit test on? Arduino, x86 (Linux/Windows), Cloud, etc?

ellisgl, 2019-04-06:
So if I where to set the mock pins for s and r to 1 and 0, then i should
run the update method, then read read the q and nq mock pins and assert
that q is 1 and nq is 0.

https://stackoverflow.com/questions/780819/how-can-i-unit-test-arduino-code

mikaelpatel, 2019-04-06:
@ellisgl 
Yes, that is one way that mock GPIO could work but you need to answer that other questions about the test harness and environment. 

Running on a x86 would require either using an AVR emulator or implementing the Arduino core in C/C++ for Linux (and/or Windows). Obviously using only GPIO is easy to mock on Linux but in most cases the test harness and test cases will use at least delay() and millis() etc. 

Next step is to write test cases for a test harness and which would use a mock GPIO class.

ellisgl, 2019-04-06:
Yeah in my case I could an emulator, since I'm running Win10. But could do
a VM or pull out my linux laptop or pull out a raspberry pi.

closed 2019-04-11 by mikaelpatel