mikaelpatel/Arduino-GPIO #13: Blink in pure c.

schleicher, 2019-07-13:
Hello.
I've tried to write Blink without Arduino IDE. Found problems:
1. LOW / HIGH are not defined. but led.low() / led.high() works ok.
2. function delay() undefined.
Is this library only to run within Arduino IDE or standalone usage is also possible?

Many Thanks

mikaelpatel, 2019-07-14:
@schleicher If I understand correctly you are asking how to use this library without the Arduino IDE? Please add information about how you are building. Also this library is C++ and will not work in pure C. There are many blogs that describe how to adapt code for more traditional build. You can find support for makefile based build in my repository Arduino-Makefile, https://github.com/mikaelpatel/Arduino-Makefile. This will include the necessary "Arduino.h". Good luck!

closed 2019-07-14 by mikaelpatel

schleicher, 2019-07-14:
Exactly so. I want to use this library outside of Arduino and preferably without Arduino dependencies. I've tested with simple Makefile. As I understand, for example delay() function is defined somewhere in Arduino.h or it's dependencies. I'm afraid that inclusion of Arduino makes my code bigger that it cannot fit in attinies.

mikaelpatel, 2019-07-14:
@schleicher Great! Then I think you know what you have to do to allow this library to be used without the Arduino core. It will take you a few hours but it is not a difficult task to write a dummy Arduino.h with the correct header includes for attiny's. Good Luck!