mikaelpatel/Arduino-GPIO #18: Not working with bad stepper drivers TB67S109AFTG

BeeMeeMan, 2023-04-06:
Hey! Use your library for my project, and all works fine with different stepper drivers.
Now bye TB67S109AFTG and nothing works.

mikaelpatel, 2023-04-08:
@BeeMeeMan Thanks for your interest in this library. Are you reporting an error or asking for help debugging your code?

BeeMeeMan, 2023-04-08:
> @BeeMeeMan Thanks for your interest in this library. Are you reporting an error or asking for help debugging your code?

Hey! Thanks for answer!. It is a really good library, thanks for it. It is greatly expands the capabilities of 16MHz. However, pin low/hight of this lib  is different from DigitalRead/Write. I rewrote this code with DigitaWrite and all works fine. I wonder if I am doing something wrong with this library(not add some parameter somewhere) or this is the price we pay for speed, or there is a bug, witch can be fixed. Issue is that the output signal on pin with this library is different from basic DigitaWrite, and some sensitive equipment may see that difference, as in my case drivers not working.

mikaelpatel, 2023-04-08:
@BeeMeeMan You need to add more info if you want help. What do you think could be useful? You are only guessing right now. And unfortunately mostly wrong. I would start by verifying that there are no symbol conflicts. The driver code might contain symbols that replace pin symbols and therefore you code uses the wrong pins. Cheers Mikael. 
PS have you measured the voltage on the pins. You might have broken your board.

BeeMeeMan, 2023-04-08:
Thanks, idea with wrong pin maybe can be good. I'll try to check it.

closed 2023-04-18 by mikaelpatel